Hoektronics/BotQueue #131: Queue choice to multiple select in "Create New Job"

StanleyT, 2015-06-19:
page: Create New Job (aka: Step 2 of 2, after choose file)
Instead of a drop down list (maybe) make a multi queue choice instead of only one queue at a time.
result: ability to send one job to several bots on several queues at one time

Jnesselr, 2015-06-23:
Not sure how I would implement it. The general design philosophy is that if you need those bots to all take the same job, then put them all in the same queue. That being said, I can see cases where that might not work well. I don't think it'll be too terrible to implement once I find a good way to present it.

Behavior wise, what would you expect the behavior to be if you submitted one job to two different queues? I'd expect that the first bot available to take the job would be the one to take it and it would be in that queue. But bots can be in multiple queues, so if you had a bot named Robert that was in both queues A and B, and you decided to submit a job overlapping queues A and B, then what queue was the job technically run in? My thoughts? both. It would basically be the intersection between the queues of the bot and the queues that you selected for the job to run in.

If you instead submitted a job overlapping queues B and C, then the job would show that it ran in queue B.

Thoughts? I could definitely see a use for having a job be able to span multiple queues.

StanleyT, 2015-07-06:
implementation:
on the Create New Job page (Step 2 of 2), instead of a drop down list, there could be check-boxes for all queues created in BotQueue (populated at time of queue creation)
-or-
on the Create New Job page (Step 2 of 2), just under the drop-down box for the queue could be an option to select another queue drop-down box (in the same fashion of the "edit bot - *** page, Queues tab)

behavior:
first bot available in first queue listed, untill first queue listed is full. moves on to next queue chosen and fills that queue, and repeats with all queues chosen untill no bots/queues are available or no jobs to assign.