Hoektronics/BotQueue #132: chronological job's list

StanleyT, 2015-06-19:
page: View Bot under Bots tab
Jobs list: list most current job first and then list in chronological order

Jnesselr, 2015-06-23:
That should be the default behavior. What does it currently do?

eric116, 2015-06-24:
Yesterday I came across a case where a job sent to G2 was numbered 1915xx. After it completed, I sent a preheat code to change the filament, and the preheat code was job 1917xx. We definitely didn't have 200 jobs sent in that time.

I believe the attached screenshot shows the jobs ordered by time, not job number. (Thought that would be the same order). Sometimes, though, there doesn't seem to be any discernible pattern at all. I'll try to get a screenshot of that case later.
[image]

Jnesselr, 2015-09-29:
This is because it's ordered on user_sort instead of on any timestamp. user_sort has had issues in the past, which is why it'll be reimplemented in 1.0. I could switch it to be a timestamp, but some of the data validation issues will cause other stuff to be a little chaotic.