Hoektronics/BotQueue #136: You cannot grab this job.

jebba, 2015-07-06:
Somehow a bot may get itself into a state where it logs to info.log this error:

[2015-07-06 15:08:48,772] ERROR: API: You cannot grab this job.
[2015-07-06 15:08:48,775] ERROR: {'api_call': 'grabjob', 'can_slice': False, 'job_id': u'202274', '_client_version': '0.5.3', '_client_name': 'Bumblebee', 'api
_output': 'json', '_uid': u'[random string]', '_local_ip': '[the ip]', 'bot_id': u'236'}

When it gets in the state, the fix is to set the job_id to 0 and it will start picking up jobs again. In the database run:

UPDATE bots SET job_id=0 WHERE NAME="BOTNAME";

For "BOTNAME" use the name of the bot, for example, for bot G1:

UPDATE bots SET job_id=0 WHERE NAME="G1";

StanleyT, 2015-07-09:
The indicators of how this issue comes up from the user side is that the job will be picked up by a bot, but instead of given the status of "taken" the status will change to "available" and be returned to jobs on deck. the above issue was caused when the bot was power cycled after accepting the job, though the job stays in "available" status

Jnesselr, 2015-08-05:
I found the location where deleting the queue would cause this issue in the old code. I still cannot for the life of me find any other location it happens in.

@StanleyT It sounds like the job is getting reset but the bot for whatever reason isn't. It might be a client and server thing, I can't tell. It looks like we're going to have better data consistency in the new code.