Hoektronics/BotQueue #141: Makerbot communication issue

wagnerN02645786, 2015-09-15:
currently i am using a replicator 2x with the serial port dev\serial\by-id\(some ID here)  and the baud rate of 115200 and i selected the makerware driver.  For some reason when i go to upload my print (i upload an x3g file sliced in makerware)  it downloads to my client, then goes to start the build (shows the temps of the extruders  and elapsed build time and percentage) , but does not preheat hangs at that page for a couple of seconds then exits to the main menu while botqueue says its printing.  I think its a communication error between bot queue decoding it to gcode or something.  If someone could help that would be great.

Jnesselr, 2015-09-15:
It doesn't decode it to gcode to send it, it's just that the file parser
had some issues. I thought I pushed that fix though. Are you running it
from github or pip?

wagnerN02645786, 2015-09-15:
i think pip i downloaded bumblebee straight from the instructions on the botqueue site

wagnerN02645786, 2015-09-15:
im running on a raspberry pi

Jnesselr, 2015-09-15:
Okay, so the main problem is I don't actually have a printer to test with. I'm currently trying to get one, and I think I'll have one in a month or so. I also just switched computers recently and haven't been able to move everything over yet.

The issue you're seeing is that I needed to convert what Makerbot gave me and change it to be slightly different, but I had to do it for the host and device commands. Currently it only works on the host part and not the device part, hence why you're seeing an issue. I haven't pushed that code yet, which means I need to go through the backup of my old laptop and find it to be able to push it.

I apologize about the current quality of BotQueue and the client. I'm currently working to remedy this situation, but it will take a bit of time.

wagnerN02645786, 2015-09-15:
thank you if you could get that working as soon as possible that would be great, im in deep need of this

Jnesselr, 2015-09-15:
Totally understandable. Could you give some more information about your time table? Because I'll do my best but I don't want to promise anything I can't deliver in time.

wagnerN02645786, 2015-09-15:
literally as soon as possible we are trying to launch our in house makerbot control system with 25 printers and multiple clients, trying to use your system.

Jnesselr, 2015-09-15:
Please shoot me a private email at admin [at] jnesselr [dot] org. Let's see
if we can't figure something out.