Hoektronics/BotQueue #142: Working Jobs references pending jobs on Queues page

StanleyT, 2015-09-29:
watching a specific queue by its direct page, the "Working Jobs" section is looking for pending jobs and does not correctly indicate a working job
[image]

Jnesselr, 2015-09-29:
What do you mean it's looking for pending jobs? Current master shows that it's grabbing jobs with the status of 'taken'. Obviously there are some bots with the status of working, so the jobs should be 'taken'. Hmm. Let's find out: (Note, I have no server I can access right now to test this query)

SELECT b.name, b.status, j.name, j.status
FROM bot b
JOIN job j
ON b.job_id = j.id
WHERE b.status='working';

This should give a list of bots that are currently working and their job status. The job status should be 'taken'. Please let me know what it is, if it isn't that.

StanleyT, 2015-09-29:
looking closer into the issue, it appears to only be affecting that one queue.  other queues are reporting correctly. oddly enough queue Y is a subset of queue -MINI- and although Y does not show the jobs correctly, -MINI- has them correctly displayed...

[image]

I can try to get the script ran if you still think it is relevant

Jnesselr, 2015-09-29:
Oh interesting. Okay, so the job is running out of mini not Y, meaning that
it won't show up in Y.

Basically, Y has no working jobs. Y has bots that are working, but they're
not working on jobs that are in Y.

StanleyT, 2015-09-29:
queue Z is a subset as well, and it is showing.... I'm trying to find out if the two sets of jobs were sent thru queue Z and queue -MINI-.

Jnesselr, 2015-09-29:
Does queue Z show Z3 and Z4?

StanleyT, 2015-09-29:
yes
and both pod Y and Z were sent through -MINI-. now to make this all more confusing pod Y started showing working jobs...

Jnesselr, 2015-09-29:
That shouldn't be possible. The job can only currently reference one queue
at a time. That's. Well... Confusing.

StanleyT, 2015-09-29:
well, good news then. currently all queues are reporting correctly. MINI is not showing any while pod Y and Z are showing their corresponding jobs.

Jnesselr, 2015-09-29:
Okay. That's good I think. If you see the issue again, please take
screenshots of the queues to show the duplication.

StanleyT, 2015-09-29:
will do

closed 2020-05-10 by Jnesselr