Hoektronics/BotQueue #143: Bumblebee crashes on Raspberry Pi / OctoPrint

bodagetta, 2016-01-21:
I tried installing Bumblebee on a raspberry pi that was set up for OctoPrint. 

It installed and ran the first time, going so far as to register with the website, but then it exits to the command line.

I've run the program several times, here is the log file

```
pi@octopi ~ $ cat /usr/local/lib/python2.7/dist-packages/bumblebee/info.log 
[2016-01-21 17:11:56,993] INFO: Started up, loading bot list.
[2016-01-21 17:11:57,018] INFO: Downloading Printrun from https://github.com/kliment/Printrun to /usr/local/lib/python2.7/dist-packages/bumblebee/engines/driver/Printrun
[2016-01-21 17:11:57,056] INFO: Downloading Printrun finished
[2016-01-21 17:13:56,377] INFO: Started up, loading bot list.
[2016-01-21 17:13:56,395] INFO: Downloading Printrun from https://github.com/kliment/Printrun to /usr/local/lib/python2.7/dist-packages/bumblebee/engines/driver/Printrun
[2016-01-21 17:13:56,430] INFO: Downloading Printrun finished
[2016-01-21 17:16:05,442] INFO: Started up, loading bot list.
[2016-01-21 17:16:05,464] INFO: Downloading Printrun from https://github.com/kliment/Printrun to /usr/local/lib/python2.7/dist-packages/bumblebee/engines/driver/Printrun
[2016-01-21 17:16:05,498] INFO: Downloading Printrun finished
[2016-01-21 17:20:47,460] INFO: Started up, loading bot list.
[2016-01-21 17:20:47,480] INFO: Downloading Printrun from https://github.com/kliment/Printrun to /usr/local/lib/python2.7/dist-packages/bumblebee/engines/driver/Printrun
[2016-01-21 17:20:47,514] INFO: Downloading Printrun finished
[2016-01-21 17:21:01,080] INFO: Started up, loading bot list.
[2016-01-21 17:21:01,103] INFO: Downloading Printrun from https://github.com/kliment/Printrun to /usr/local/lib/python2.7/dist-packages/bumblebee/engines/driver/Printrun
[2016-01-21 17:21:01,138] INFO: Downloading Printrun finished
[2016-01-21 17:28:16,378] INFO: Started up, loading bot list.
[2016-01-21 17:28:16,398] INFO: Downloading Printrun from https://github.com/kliment/Printrun to /usr/local/lib/python2.7/dist-packages/bumblebee/engines/driver/Printrun
[2016-01-21 17:28:16,433] INFO: Downloading Printrun finished
pi@octopi ~ $ ```

Jnesselr, 2016-01-21:
I'm not sure how Bumblebee runs on octopi, nor am I sure what the effects of installing bumblebee with root is. I'm presuming based on the path you installed bumblebee with root, correct? It's actually advised not to do it that way, but instead install it with a virtualenv.

The commands for that are (Putting the virtual env in your current directory, I'm trying to make it so that installing with root automatically does this):
```bash
$ sudo pip install virtualenv
$ virtualenv .env
$ source .env/bin/activate
$ pip install bqclient
```

As far as I can tell, your issue stems from the fact that Printrun can't actually download because it was installed as root.

bodagetta, 2016-01-21:
You're right, I did have to install it as root. I had permission errors otherwise.

I uninstalled and re-installed in the virtual environment.

I was able to register the Pi again, and I got a screen about scanning, then it exited again. When I re-run Bumblebee it flashes that gray screen then back to the command line.

```
(.env)pi@octopi ~/.env/lib/python2.7/site-packages/bumblebee $ cat info.log
[2016-01-21 18:03:14,313] INFO: Started up, loading bot list.
[2016-01-21 18:03:14,341] INFO: Downloading Printrun from https://github.com/kliment/Printrun to /home/pi/.env/lib/python2.7/site-packages/bumblebee/engines/driver/Printrun
[2016-01-21 18:03:43,347] INFO: Downloading Printrun finished
[2016-01-21 18:04:33,095] INFO: Started up, loading bot list.
[2016-01-21 18:06:05,070] INFO: Started up, loading bot list.
(.env)pi@octopi ~/.env/lib/python2.7/site-packages/bumblebee $ 
```
is the new log file. Thanks for your help!

Jnesselr, 2016-01-29:
Can you run pip freeze? I might have to ask you to edit a file to see the actual error. I made a decision a while back that's come to bite me every time something like this happens.

bodagetta, 2016-02-01:
Is this what you need?

```
pi@octopi ~/.env/lib/python2.7/site-packages/bumblebee $ pip freeze
Warning: cannot find svn location for distribute==0.6.24dev-r0
Pygments==2.1
RPi.GPIO==0.5.11
argparse==1.2.1
beautifulsoup4==4.4.1
## FIXME: could not find svn URL in dependency_links for this package:
distribute==0.6.24dev-r0
makerbot-driver==0.1.3
netifaces==0.8
numpy==1.6.2
picamera==1.10
pifacecommon==4.1.2
pifacedigitalio==3.0.4
pygame==1.9.1release
pygobject==3.8.2
pyserial==3.0.1
requests==2.9.1
requests-oauth==0.4.1
virtualenv==1.7.1.2
wsgiref==0.1.2
pi@octopi ~/.env/lib/python2.7/site-packages/bumblebee $ 
```
```

bodagetta, 2016-02-08:
I just realized you probably wanted me to run pip freeze inside the virtual environment 

```
(.env)pi@octopi ~ $ pip freeze
Pygments==2.1
argparse==1.2.1
beautifulsoup4==4.4.1
bqclient==0.5.5
distribute==0.6.24
makerbot-driver==0.1.3
pyserial==3.0.1
requests==2.9.1
requests-oauth==0.4.1
wsgiref==0.1.2
(.env)pi@octopi ~ $ 
```

bodagetta, 2016-02-08:
I've put some logging lines into the code to try to figure out where it is crashing. It seems like it is crashing in the scanDevices function at the `scanData = json.dumps(data)` line. 

I tried the json.dumps command in a python console and didn't get any errors.

Could it have something to do with how the ports are formatted when retrieved from hive.scanBots()?

```
    def scanDevices(self):

        # look up our data
        data = {}
        self.log.info("looking up bots")
        data['bots'] = hive.scanBots()
        self.log.info("looking up cameras")
        data['cameras'] = camera_control.scanCameras()
        self.log.info(data)
        scanData = json.dumps(data)
        self.log.info("json dumps completed")
        self.log.info(scanData)
        if scanData != self.lastScanData or (self.lastImageTime + 60 < time.time()):
            self.lastScanData = scanData

            camera_files = []
            if len(data['cameras']):
                for idx, camera in enumerate(data['cameras']):
                    outfile = camera['name'] + '.jpg'
                    try:
                        if camera_control.takePicture(camera['device'], watermark=None, output=outfile):
                            self.lastImageTime = time.time()
                            fullImgPath = hive.getImageDirectory(outfile)
                            camera_files.append(fullImgPath)
                    except Exception as ex:
                        self.log.exception(ex)

            # now update the main site
            self.api.sendDeviceScanResults(data, camera_files)
```

```
[2016-02-08 18:57:31,005] INFO: Started up, loading bot list.
[2016-02-08 18:57:31,007] INFO: sys.stdout.isatty
[2016-02-08 18:57:31,016] INFO: trying screen
[2016-02-08 18:57:31,018] INFO: show intro
[2016-02-08 18:57:31,024] INFO: after refresh
[2016-02-08 18:57:31,026] INFO: before updates
[2016-02-08 18:57:31,027] INFO: checking messages
[2016-02-08 18:57:31,027] INFO: done checking messages
[2016-02-08 18:57:31,028] INFO: getting bots
[2016-02-08 18:57:31,029] INFO: scanning devices
[2016-02-08 18:57:31,030] INFO: looking up bots
[2016-02-08 18:57:31,133] INFO: looking up cameras
[2016-02-08 18:57:31,164] INFO: {'bots': {'printcoredriver': [<serial.tools.list_ports_linux.SysFS object at 0xd83770>, <serial.tools.list_ports_linux.SysFS object at 0xd83850>]}, 'cameras': []}
```

Jnesselr, 2016-02-08:
I know what the issue is. Install pyserial 2.7 instead of 3.0.1. They
changed how that function returns values. I thought I pushed that code up,
but it seems I'm still working on refactoring that part.

"pip install pyserial==2.7" should work inside the virtual environment.

bodagetta, 2016-02-09:
That got it to load!

I'm able to add a bot via the web interface and it shows the correct port in the dropdown (/dev/ttyACM0). 

I created a job and uploaded it and assigned it. I can see it is being assigned in the bumblebee interface but it just runs to 100% without controlling the printer at all.

I'm uploading GCODE. 

Any ideas?

Jnesselr, 2016-02-09:
Can you show the last ~100 or so lines of your info.log file? If you don't know where that is, use the command "bumblebee log" inside the virtual environment, and it will print out the path.

bodagetta, 2016-02-10:
Sure - here you go

```
[2016-02-09 21:45:27,662] INFO: Started up, loading bot list.
[2016-02-09 21:45:27,689] INFO: Downloading Printrun from https://github.com/kliment/Printrun to /home/pi/.env/lib/python2.7/site-packages/bumblebee/engines/driver/Printrun
[2016-02-09 21:45:58,242] INFO: Downloading Printrun finished
[2016-02-09 22:01:33,736] INFO: Shutting down.
[2016-02-09 22:01:47,683] INFO: Started up, loading bot list.
[2016-02-09 22:01:49,639] INFO: Creating worker thread for bot Lulzbot Mini
[2016-02-09 22:01:49,642] INFO: Loading bot Lulzbot Mini
[2016-02-09 22:01:49,652] INFO: Lulzbot Mini: Bot startup
[2016-02-09 22:02:00,807] INFO: Lulzbot Mini: Changing status from offline to idle
[2016-02-09 22:05:23,456] INFO: Shutting down.
[2016-02-09 22:05:23,473] INFO: Lulzbot Mini: Shutting down.
[2016-02-09 22:05:23,475] DEBUG: Lulzbot Mini: Exiting.
[2016-02-09 22:07:56,556] INFO: Started up, loading bot list.
[2016-02-09 22:08:02,752] INFO: Creating worker thread for bot Lulzbot Mini
[2016-02-09 22:08:02,756] INFO: Loading bot Lulzbot Mini
[2016-02-09 22:08:02,776] INFO: Lulzbot Mini: Bot startup
[2016-02-09 22:18:15,245] INFO: Lulzbot Mini: Changing status from idle to working
[2016-02-09 22:18:15,257] INFO: Downloading https://botqueue.s3.amazonaws.com/assets/f4/d8/2f/2b/3c/gears.gcode to /home/pi/.env/lib/python2.7/site-packages/bumblebee/cache/3e26b91d79db6931ae8ba3891abbf7eb-gears.gcode
[2016-02-09 22:18:26,246] INFO: Download complete: https://botqueue.s3.amazonaws.com/assets/f4/d8/2f/2b/3c/gears.gcode
[2016-02-09 22:18:28,147] DEBUG: Waiting for driver to connect.
[2016-02-09 22:19:05,644] DEBUG: Bumbledrive: print thread entry.
[2016-02-09 22:19:05,996] INFO: Lulzbot Mini: print: 0.00%
[2016-02-09 22:19:06,948] DEBUG: Bumbledrive: finishing print.
[2016-02-09 22:19:07,285] INFO: Lulzbot Mini: Print finished.
[2016-02-09 22:19:07,287] INFO: Lulzbot Mini: print: 0.00%
[2016-02-09 22:19:07,747] DEBUG: Lulzbot Mini: Bot finished @ state waiting
[2016-02-09 22:19:21,629] INFO: Mothership: Lulzbot Mini status changed from working to waiting
[2016-02-09 22:19:21,632] DEBUG: Worker for Lulzbot Mini is stale: 1455056361.43 / 1455056361.63
[2016-02-09 22:20:50,971] INFO: Lulzbot Mini: Changing status from waiting to idle
[2016-02-09 22:21:14,520] INFO: Lulzbot Mini: Changing status from idle to working
[2016-02-09 22:21:14,616] INFO: Using cached file /home/pi/.env/lib/python2.7/site-packages/bumblebee/cache/3e26b91d79db6931ae8ba3891abbf7eb-gears.gcode
[2016-02-09 22:21:16,445] DEBUG: Waiting for driver to connect.
[2016-02-09 22:21:17,447] DEBUG: Waiting for driver to connect.
[2016-02-09 22:21:18,449] DEBUG: Waiting for driver to connect.
[2016-02-09 22:21:19,451] DEBUG: Waiting for driver to connect.
[2016-02-09 22:21:59,475] DEBUG: Bumbledrive: print thread entry.
[2016-02-09 22:21:59,835] INFO: Lulzbot Mini: print: 0.00%
[2016-02-09 22:22:00,634] DEBUG: Bumbledrive: finishing print.
[2016-02-09 22:22:01,513] INFO: Lulzbot Mini: Print finished.
[2016-02-09 22:22:01,515] INFO: Lulzbot Mini: print: 0.00%
[2016-02-09 22:22:02,009] DEBUG: Lulzbot Mini: Bot finished @ state waiting
[2016-02-09 22:22:02,219] INFO: Mothership: Lulzbot Mini status changed from working to waiting
[2016-02-09 22:22:23,495] INFO: Bot Lulzbot Mini no longer found, shutting it down.
[2016-02-09 22:22:23,619] INFO: Lulzbot Mini: Shutting down.
[2016-02-09 22:22:23,621] DEBUG: Lulzbot Mini: Exiting.
[2016-02-09 22:23:19,134] INFO: Creating worker thread for bot Lulzbot Mini #1
[2016-02-09 22:23:19,138] INFO: Loading bot Lulzbot Mini #1
[2016-02-09 22:23:19,147] INFO: Lulzbot Mini #1: Bot startup
[2016-02-09 22:23:30,562] INFO: Lulzbot Mini #1: Changing status from offline to idle
[2016-02-09 22:27:19,101] INFO: Lulzbot Mini #1: Changing status from idle to working
[2016-02-09 22:27:19,198] INFO: Using cached file /home/pi/.env/lib/python2.7/site-packages/bumblebee/cache/3e26b91d79db6931ae8ba3891abbf7eb-gears.gcode
[2016-02-09 22:27:20,967] DEBUG: Waiting for driver to connect.
[2016-02-09 22:28:00,216] DEBUG: Bumbledrive: print thread entry.
[2016-02-09 22:28:00,417] INFO: Lulzbot Mini #1: print: 0.00%
[2016-02-09 22:28:01,220] DEBUG: Bumbledrive: finishing print.
[2016-02-09 22:28:01,642] INFO: Lulzbot Mini #1: Print finished.
[2016-02-09 22:28:01,649] INFO: Lulzbot Mini #1: print: 0.00%
[2016-02-09 22:28:02,050] DEBUG: Lulzbot Mini #1: Bot finished @ state waiting
[2016-02-09 22:28:02,167] INFO: Mothership: Lulzbot Mini #1 status changed from working to waiting
[2016-02-09 22:28:40,194] INFO: Shutting down.
[2016-02-09 22:28:40,646] INFO: Lulzbot Mini #1: Shutting down.
[2016-02-09 22:28:40,648] DEBUG: Lulzbot Mini #1: Exiting.
```