Hoektronics/BotQueue #145: Job available not taken

wagnerN02645786, 2016-03-10:
Botqueue will mark the job as taken, but on the bumblebee client the job will still be marked available and the job will not print.

Please help,

Thanks,
Mitchell

HuongAT, 2016-03-29:
you should look at  https://github.com/Hoektronics/bumblebee/issues/8

Jnesselr, 2016-03-29:
Actually, I'm fairly certain they're different. Your issue happens because the driver crashes and calls the finally code.

If that were the case, the job would be considered complete even though it threw an error.

wagnerN02645786, 2016-04-07:
According to Botqueue the job is marked taken and will continue to say working forever unless i do something.  Bumblebee on the other hand will mark the job as available and will not try to communicate at all with the printer.  by restarting bumblebee this fixes the problem but i would like to see if there is a better fix for this.

Thanks

Jnesselr, 2016-04-07:
Can you upload bumblebee's log file please?

wagnerN02645786, 2016-04-07:
I've haven't had an instance of this happening recently it would take me forever to go through the log to find it.  ill upload when it happens again.  i have cleared the log so it would be easier to find

wagnerN02645786, 2016-04-11:
Ok so it happened again, this time on one of my replicator 2x's   here is my log file.  The Printers name is Rep 2x #2.  The problem build starts to occur at around: [2016-04-11 16:32:21,484].  and the problem itself is at: [2016-04-11 16:45:21,334].  start there and keep reading.
[info.log.txt](https://github.com/Hoektronics/BotQueue/files/213891/info.log.txt)