Hoektronics/BotQueue #157: I followed the setup exact and bumblebee is nowhere to be found

benbeezy, 2018-07-07:
I set this up on a new raspberry pi and followed the instructions to a T and everything looks like it is working till I get down to the last line of starting bumblebee. It just tells me that it has no idea what I am talking about. Am I missing something?

Jnesselr, 2018-07-08:
Just a heads up, I'm currently rewriting this software as a version 2, so my support for version 1 (the current version) is limited. That being said, which instructions did you follow? I can try to run through the instructions myself on a fresh PI and see if there's an issue. For now, there's not too much in this issue to go on.

benbeezy, 2018-07-08:
I was following this https://www.botqueue.com/help on the raspberry pi tab.
But bumblebee was never installed as far as I can tell. Whereis bumblebee
returns nothing.

Jnesselr, 2018-10-05:
Just to update this thread, so I don't lose track of it (after months whoops) the issue you're seeing is that the bumblebee binary doesn't get installed anyway on your path. For my test, it was installed at "/home/pi/.local/bin"

There is another issue, which is that printcore and pyserial are now py3 only. Which is great news for BotQueue v2, but terrible news if you're trying to use BotQueue v1.

closed 2020-05-10 by Jnesselr